MicrosoftDocs/windows-driver-docs #777: Command List?

bitcrazed, 2018-08-20:
It'd be great to have all the debugger commands listed here, each linked to their respective page.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b2e86f45-ca76-4b44-b5c9-f62c742e4dce
* Version Independent ID: 22ff867d-dac6-7876-4fad-57eb8ef3b378
* Content: [Commands](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/commands)
* Content Source: [windows-driver-docs-pr/debugger/commands.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/commands.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

wdkbot, 2018-08-20:
Hi @bitcrazed, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @DOMARS to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-08-20:
Thanks, @bitcrazed 
@DOMARS I think that this could be done easily by referencing the TOC file. Can you scope it out and prioritize the work?
@bitcrazed you could take a shot at it and create a commit if you're so inclined. :-)

DOMARS, 2018-08-20:
@bitcrazed - Thanks for the input. I'm going to check with the tools team to see if we can automate displaying the list of subtopics that are in the TOC to the left on this page.

bitcrazed, 2018-08-20:
Thanks @DOMARS - I think that'd be a much better solution vs. a point-fix.

DOMARS, 2019-09-11:
@bitcrazed - Thanks for the input. As our team is pretty resource constrained, it does not look that we will be able to pursue the automation required to create a list of debugger commands. I have had conversations with the dev team about automating testing of specific commands that would be tied to the docs, but that is not likely to get resourced any time soon. This idea will carry on, and maybe at some point via a hackathon or some other avenue, it will re-emerge in the future.  Thanks again for the idea and your interest in improving the debugger docs.

closed 2019-09-11 by DOMARS