MicrosoftDocs/windows-driver-docs #779: VHD vs. VHD file

ManuelSchenkel, 2018-08-21:
Regarding the part about UNMAP request:
What's the difference between VHD and VHD File? Isn't every VHD a VHD file?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b67b2f1c-add1-0e22-29d1-8fe933af13c9
* Version Independent ID: 0ae1ca0f-68c6-5324-277d-d9d141eba1d5
* Content: [Thin Provisioning](https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/thin-provisioning)
* Content Source: [windows-driver-docs-pr/storage/thin-provisioning.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/storage/thin-provisioning.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-08-21:
Hi @ManuelSchenkel, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @EliotSeattle to this issue, and they'll triage this as soon as they have a moment.

lorihollasch, 2019-10-03:
@ManuelSchenkel , our apologies for the LONG time it took to get to your issue! I've updated the page to be clear about VHD (which can be VHD file or VHDX file). Thanks for reporting this.

closed 2019-10-03 by lorihollasch