MicrosoftDocs/windows-driver-docs #780: ProductName & productName in Request Body example

GabeSJones, 2018-08-21:
The first example has both ProductName and productName fields.  I think only the latter should be there.  Also, TestHarness should probably be testHarness to match the documentation at https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-product-data#product-resource

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 24518ab0-bd10-1a04-f3a0-1ba754409841
* Version Independent ID: ad632005-a43b-8d81-c4ef-8730d59b4363
* Content: [Create a new product](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/create-a-new-product)
* Content Source: [windows-driver-docs-pr/dashboard/create-a-new-product.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/create-a-new-product.md)
* Product: **windows-hardware**
* GitHub Login: @balapv
* Microsoft Alias: **balapv**

wdkbot, 2018-08-21:
Hi @GabeSJones, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @EliotSeattle to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-12-05:
Sorry, @GabeSJones. I somehow missed these. I made the corrections requested in this Issue, and I am checking with the product team on the HLK/Attestation combination as well as the existence of the mythical packageType.

closed 2018-12-05 by EliotSeattle