MicrosoftDocs/windows-driver-docs #781: Also, packageType and TestHarness

GabeSJones, 2018-08-21:
Also,   "TestHarness": "Attestation" and   "packageType": "HLK".  Those two seem to be at odds with one another, and there is no "packageType" documented on the Product Resource page.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 24518ab0-bd10-1a04-f3a0-1ba754409841
* Version Independent ID: ad632005-a43b-8d81-c4ef-8730d59b4363
* Content: [Create a new product](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/create-a-new-product)
* Content Source: [windows-driver-docs-pr/dashboard/create-a-new-product.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/create-a-new-product.md)
* Product: **windows-hardware**
* GitHub Login: @balapv
* Microsoft Alias: **balapv**

wdkbot, 2018-08-21:
Hi @GabeSJones, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @EliotSeattle to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-12-05:
Apologies for not seeing this request. I have sent a request to the product team to get some answer for these two questions.

Matsogi, 2019-11-20:
Hello, can we expect some answers? I have troubles with creating new product (caused by error 422 - Unprocessable Entity) I thing it is related to this. Could you please clarify and fix this documentation? Thank You in advance.

tedhudek, 2020-11-03:
I inherited this issue and just started investigating, with no experience in Partner center.  My notes:

On https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/create-a-new-product, in the example JSON
request body for creating a new product, are both  "TestHarness": "Attestation" and "packageType": "HLK" required?

https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-product-data#product-resource does not document packageType, but it does document testHarness (leading lower case).

Is the correct solution to change TestHarness to testHarness on https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/create-a-new-product and remove the packageType line?

Investigating but if the community has info that would help.

tedhudek, 2020-11-03:
Confirmed.  The updated doc should be live within a few minutes.  Thanks for the heads-up and sorry it took us a while!

closed 2020-11-03 by tedhudek