MicrosoftDocs/windows-driver-docs #782: Page uses wikipedia content without sharing as CC

karel-3d, 2018-08-22:
This page seems to be copying parts of wikipedia article without acknowledging that, which is violation of Wikipedia copyright.

I have originally thought that Wikipedia copied this article, but it seems when looking at the history - https://en.wikipedia.org/w/index.php?title=Human_interface_device&diff=next&oldid=480078218 - the wikipedia article used to use slightly different sentences before settling for the ones used in the current article state (and the text here).

For example, wikipedia sentence

&gt; Prior to the introduction of the HID concept, devices usually conformed to strictly defined protocols for mouse, keyboards and joysticks; for example, the standard mouse-protocol at the time supported relative X- and Y-axis data and binary input for up to two buttons, with no legacy support. 

is copied here, as is the next sentence (almost)

&gt; All hardware innovations necessitated either overloading the use of data in an existing protocol or the creation of custom device drivers and the evangelization of a new protocol to developers.

(it is slightly changed here, maybe older version of WP article)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9bf1b7f1-a59a-d9ef-7c8e-b994946afb32
* Version Independent ID: 3fcf10e3-c148-702e-9d9e-01bd87136d61
* Content: [Introduction to HID Concepts](https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/introduction-to-hid-concepts)
* Content Source: [windows-driver-docs-pr/hid/introduction-to-hid-concepts.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/hid/introduction-to-hid-concepts.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-08-22:
Hi @karel-3d, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @PageWriter-MSFT to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-08-22:
Thanks, @karel-3d for bringing this to our attention. @PageWriter-MSFT will look into the topic and do the right thing.

PageWriter-MSFT, 2018-08-22:
Hi @karel-3d . Thanks for reporting this issue! We try to do our best. This was not it :( The description has been removed.

closed 2018-08-22 by PageWriter-MSFT