MicrosoftDocs/windows-driver-docs #783: Exception codes list

LuigiBruno, 2018-08-23:
Please provide a more complete and exhaustive list of the exception codes that could be displayed as values for the parameter 1.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 06210c17-833c-57fb-d0bf-e899cc9cc48c
* Version Independent ID: 05435dcd-4026-8423-cebd-4e60996253bf
* Content: [Bug Check 0x1E KMODE_EXCEPTION_NOT_HANDLED](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1e--kmode-exception-not-handled)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1e--kmode-exception-not-handled.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1e--kmode-exception-not-handled.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

wdkbot, 2018-08-23:
Hi @LuigiBruno, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @DOMARS to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-08-23:
Thanks, @LuigiBruno for the request.
@DOMARS can you please triage and prioritize this in relation to your other work for our fine MVP friend?

DOMARS, 2018-08-23:
@LuigiBruno - Thanks for the request to add more information on the exception codes. I added a link to the NT Status Values topics that lists all of them. The changes should be live within 24 hours. Thanks for your ongoing interest in improving our stop code documentation.

closed 2018-08-23 by DOMARS

LuigiBruno, 2018-09-29:
Thanks @EliotSeattle and @DOMARS.