MicrosoftDocs/windows-driver-docs #786: Link Broken?

csm10495, 2018-08-24:
I think the IOCTLs link (https://msdn.microsoft.com/library/windows/hardware/ff556290#wdkgloss-ioctl) isn't actually going to the correct place as it routes to a generic page instead of something on IOCTLs.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4076dd23-433e-2ec1-789a-e70b2e447fb7
* Version Independent ID: c9a5dc47-62e4-bdb9-b28d-aa1d6d0c78d1
* Content: [IRP_MJ_DEVICE_CONTROL](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/irp-mj-device-control)
* Content Source: [windows-driver-docs-pr/kernel/irp-mj-device-control.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/irp-mj-device-control.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-08-24:
Hi @csm10495, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @PageWriter-MSFT to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-08-24:
Fixed in source. Waiting for it to go live.

"Every driver whose device objects belong to a particular device type (see [Specifying Device Types](https://msdn.microsoft.com/library/windows/hardware/ff563821) is required to support this request in a [DispatchDeviceControl](https://msdn.microsoft.com/library/windows/hardware/ff543287) routine, if a set of system-defined I/O control codes (IOCTLs) exists for the type. For more info about IOCTLs, see [Introduction to I/O Control Codes](https://msdn.microsoft.com/library/windows/hardware/ff556290#wdkgloss-ioctl)."

closed 2018-08-24 by EliotSeattle