MicrosoftDocs/windows-driver-docs #788: Generating the .cab files

haephrati, 2018-08-24:
I used the method described in this article https://www.codeproject.com/Articles/1257731/Creating-a-cab-archive-from-one-or-more-files

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ed287852-04b7-1569-fd51-2640cfa24f57
* Version Independent ID: a8ce732f-5b41-f1ce-e37d-a7c2794546a4
* Content: [Attestation signing a kernel driver for public release](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/attestation-signing-a-kernel-driver-for-public-release)
* Content Source: [windows-driver-docs-pr/dashboard/attestation-signing-a-kernel-driver-for-public-release.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/attestation-signing-a-kernel-driver-for-public-release.md)
* Product: **windows-hardware**
* GitHub Login: @aahill
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-08-24:
Hi @haephrati, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @EliotSeattle to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-08-25:
Hi @haephrati 
Is there an issue about the documentation? I can't tell from what's captured in this form.
Thanks!

haephrati, 2018-08-25:
No. No issue at all. Only meant to comment.

EliotSeattle, 2018-08-27:
Well, then, thank you for the comment!

closed 2018-08-27 by EliotSeattle