MicrosoftDocs/windows-driver-docs #789: FindExtensionNode Fail

qinhaicheng, 2018-08-27:
FindExtensionNode Fail，When the driver is loaded

---
#### 文档详细信息

⚠ *请勿编辑此部分。 docs.microsoft.com ➟ GitHub 问题链接必须具有此部分。*

* ID: 6767e28b-a1ec-00a8-da87-f70eb60c6384
* Version Independent ID: c43f6c76-5abe-8fd9-3bb8-2fb14a5419a3
* Content: [Sample Application for UVC Extension Units](https://docs.microsoft.com/zh-cn/windows-hardware/drivers/stream/sample-application-for-uvc-extension-units)
* Content Source: [windows-driver-docs-pr/stream/sample-application-for-uvc-extension-units.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/stream/sample-application-for-uvc-extension-units.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-08-27:
Hi @qinhaicheng, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @barrygolden to this issue, and they'll triage this as soon as they have a moment.

qin-HC, 2018-08-29:
搞定了，可以不需要安装驱动，直接在应用层就可以调用了。刚开始按照这个文档做，被带坑里了！(╥╯^╰╥)

EliotSeattle, 2018-08-29:
Thank you @qin-HC 
@barrygolden can you see if you can find someone to test the sample code?

sun-xm, 2019-03-15:
> 搞定了，可以不需要安装驱动，直接在应用层就可以调用了。刚开始按照这个文档做，被带坑里了！(╥╯^╰╥)

怎么搞定得？能给个sample瞅瞅不？谢了 @qin-HC

qin-HC, 2019-04-27:
You can refer to https://github.com/3dtof/voxelsdk/blob/master/Voxel/UVCXU.cpp

sun-xm, 2019-04-28:
> You can refer to https://github.com/3dtof/voxelsdk/blob/master/Voxel/UVCXU.cpp

非常感谢！

barrygolden, 2021-01-27:
Thanks for reaching out to the Windows Driver Developer Documentation team.  You may have success posting your question about the FindExtensionNode function failure you are experiencing in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

[Hardware and Application Developers on Q&A camera hardware forum](https://docs.microsoft.com/answers/topics/windows-hardware-code-camera.html)

[NTDEV forum at OSR Online](https://www.osronline.com/showlists.cfm?list=ntdev)

closed 2021-01-27 by barrygolden