MicrosoftDocs/windows-driver-docs #791: Wrong location given for documentation

assarbad, 2018-08-28:
The article gives this (emphasis mine):

> Also, refer to the readme file Selfsign_readme.htm and the script file Selfsign_example.cmd, which are located in the **src\general\build\driversigning** directory of the Windows Driver Kit (WDK). **The script file Selfsign_example.cmd** includes a documented example of how to use MakeCert to create a test certificate.

Two issues here. In the modern Windows Kits the HTML file seems to be in a different location. I found two:

* `C:\Program Files (x86)\Windows Kits\10\bin\selfsign`
* `C:\Program Files (x86)\Windows Kits\10\bin\10.0.17134.0\selfsign`

However, this may be owed to the fact that I only have the older self-contained WDKs (and DDKs) alongside the 10.0.17134.0 one, but just this single "modern" WDK.

The mentioned `.cmd` script does not seem to exist at all.

Side-note: the older self-contained **Vista** WDK actually has it in the same location as the "modern" WDKs (relative to the WDK base directory) but the Windows 7 SP1 WDK has it in the location mentioned in the article as of this writing.

```
C:\WINDDK\6001.18002\bin\SelfSign\selfsign_readme.htm
C:\WINDDK\6001.18002\bin\SelfSign\selfsign_example.cmd
C:\WINDDK\7600.16385.1\src\general\build\driversigning\selfsign_readme.htm
C:\WINDDK\7600.16385.1\src\general\build\driversigning\selfsign_example.cmd
```
### Furthermore a few inconsistencies

This article, which precedes [this one](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/makecert-test-certificate) in the TOC pane is subtly different from the latter. For example the name of the certificate differs in the invocation given.

Also it's not clear why there have to be two nearly identical articles regarding how to create a test certificate with `makecert.exe`.

The file naming inconsistencies in example invocations stretch across the whole section for the test signing topic.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 33b25f0f-6898-0e7a-ae15-cf609e167a77
* Version Independent ID: b0907ff5-7e18-5dfa-72d5-02c9be59691b
* Content: [Creating Test Certificates](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/creating-test-certificates)
* Content Source: [windows-driver-docs-pr/install/creating-test-certificates.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/creating-test-certificates.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-08-28:
Hi @assarbad, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @tedhudek to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-08-28:
Wow, thanks, @assarbad for taking the time to go through this and provide this detailed feedback.
@tedhudek can you please take a look at this one?

tedhudek, 2018-09-07:
Hi @assarbad, thanks for the feedback, and apologies for the delay. I have made some changes to the documentation to address your suggestions. Specifically:
 
 * updated  https://docs.microsoft.com/windows-hardware/drivers/install/creating-test-certificates with the correct Windows 10 path for the selfsign htm
 * removed the .cmd reference since it no longer exists
 * consolidated info that had been duplicated in https://docs.microsoft.com/windows-hardware/drivers/install/makecert-test-certificate
 
 I also have a resolution for #792 and will update that issue momentarily.

closed 2018-09-07 by tedhudek