MicrosoftDocs/windows-driver-docs #792: Wouldn't it be better to limit the test-signing certificate usage?

assarbad, 2018-08-28:
Hi,

the example invocations of `makecert.exe` don't use the command line switch `-eku 1.3.6.1.5.5.7.3.3` to limit the usage of the resulting certificate to code-signing. Wouldn't it be better to recommend those certificates to be created with the bare minimum of EKUs necessary? As it stands the result is an all-purpose certificate.

And while I understand and appreciate that the certificate is only relevant for in-house use, it seems odd that the [ELAM sample README](https://github.com/Microsoft/Windows-driver-samples/blob/master/security/elam/README.md) mentions:

> Early Launch drivers are required to be signed with a code-signing certificate that also contains the Early Launch EKU "1.3.6.1.4.1.311.61.4.1". In a production environment, Early Launch drivers are signed by Microsoft for qualifying Anti-Malware vendors with a WHQL certificate that contains this EKU. The makecert.exe tool can be used to generate a self-signed test certificate that contains both the Early Launch EKU and the "1.3.6.1.5.5.7.3.3" Code Signing EKU. Once a certificate of this form has been created, signtool.exe can be used to sign elamsample.sys.

... thus recommending the bare minimum necessary.

Best regards,

Oliver
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 33b25f0f-6898-0e7a-ae15-cf609e167a77
* Version Independent ID: b0907ff5-7e18-5dfa-72d5-02c9be59691b
* Content: [Creating Test Certificates](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/creating-test-certificates)
* Content Source: [windows-driver-docs-pr/install/creating-test-certificates.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/creating-test-certificates.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-08-28:
Hi @assarbad, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @tedhudek to this issue, and they'll triage this as soon as they have a moment.

tedhudek, 2018-09-07:
Thanks @assarbad, for the suggestion.  You are correct, it is good practice to use the explicit list of ekus. I have updated https://docs.microsoft.com/windows-hardware/drivers/install/creating-test-certificates accordingly.  Please don't hesitate if you have additional doc feedback. Also note that you can click **Edit** from any page on docs.microsoft.com if you want to start a PR with your own proposed changes.

closed 2018-09-07 by tedhudek

assarbad, 2018-09-14:
Oh, alright, I'll probably give those pull requests a chance next time around. Thanks for your stellar work in keeping things polished!