MicrosoftDocs/windows-driver-docs #812: Apostrophe not showing

laszloC, 2018-08-29:
The apostrophe in "driver's" is shown coded, not rendered, as &#39; instead of '

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 745bf994-7ea9-016c-c4f4-bab0173195eb
* Version Independent ID: a0c4a4d7-1001-949f-f9d2-22448938c219
* Content: [Specifying an Unload Function](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/specifying-an-unload-function)
* Content Source: [windows-driver-docs-pr/network/specifying-an-unload-function.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/specifying-an-unload-function.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-08-29:
Hi @laszloC, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @duncanmacmichael to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-08-29:
Hi @laszloC 
Thanks for the feedback. I'm not sure what you're seeing, though. I see this, which looks correct. Am I missing your point?
[image]

EliotSeattle, 2018-08-29:
Oh yeah. That's ugly. I changed those instances and pushed to master. Will go live the next time we synch our internal repo with the live site. Thanks for catching that! In the future, feel free to make an edit and commit the change. Cheers!

closed 2018-08-29 by EliotSeattle