MicrosoftDocs/windows-driver-docs #817: "Unable to read dynamic function table entry at" spam

sebas77, 2018-08-31:
Can I disable "Unable to read dynamic function table entry at" in WinDbg Preview?
Its continous spam is preventing me from debugging an application. I would like to know why it happens to and how to prevent it.

EliotSeattle, 2018-08-31:
Thanks, @sebas77. Looking around, it seems like you are not the only one  wanting to be able to sidestep this. The first thing I'd say, and this is iterated on the various posts and discussions, is that this is a security feature. Toggle it at your own risk.
@DOMARS, can you please look into whether we should advise how to turn this off and how? And see if it rises in priority against your other deliverables. I found a couple of names and ideas in this thread: https://github.com/dotnet/coreclr/issues/4402.

sebas77, 2018-09-10:
Hello, I am back in office, I noticed it has been marked as in-progress, what does it mean?

DOMARS, 2018-09-15:
@sebas77 - Typically "in-progress" means that I have spent some time looking at this issue. 

As you may have seen, one solution is to disable the security feature that keeps the debugger from only using known stack walkers as discussed in this link.
https://stackoverflow.com/questions/39255930/can-i-disable-the-unable-to-read-dynamic-function-table-entry-message-in-windb

But be aware that the turning this option on means that the debugger might be able to execute code that is not properly checked, so only use this option as required on a test PC, not connected to the network as it opens up a security hole. 

I can't speak to which element in your dev environment is causing this, I would assume by loading or causing to load, a custom stack walker. This issue provided more specific information and describes a process of white listing, that might be an option to investigate. 

https://github.com/dotnet/coreclr/issues/4402

I'm not sure of all of the conditions that out of stack walkers can be loaded and which elements such as unity would be most likely to do this. That may be a bit of a project to track down - the component vendors may have insight into this. 

One resource for technical questions are the online forums. For example, these lists are pretty responsive.

[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
[OSR's ntdev List] (https://www.osronline.com/showlists.cfm?list=ntdev)
[OSR's winbbg List]http://www.osronline.com/cf.cfm?PageURL=showlists.cfm?list=WINDBG

Thanks for this input and I will let you know if the dev team is able to provide addtional information, and if they do, I will re-open this issue.  Good luck in your work to track this down.

closed 2018-09-15 by DOMARS

sebas77, 2018-09-15:
Thanks for the reply and the clue you gave, it can be useful. I just want to add that the stackoverflow solution didn't work for me on windbg preview, I actually would love to know if I am right or I did something wrong. I didn't try to change the registry though.