MicrosoftDocs/windows-driver-docs #818: Please fix "doesn't have any references to other projects and no projects were packaged." error.

ahidaka, 2018-08-31:
I have several WDK development machines. And test to make simple WinUSB driver package as this way many times. Some machines are sometime / always fail to build WinUSB package. 

There happens the same error as stackoverflow and OSR articles as below.

https://stackoverflow.com/questions/22574095/windows-usb-tutorial
https://www.osronline.com/showthread.cfm?link=273883

But my machines made fail as below. And no packages were built.

1&gt;Stamping [Version] section with DriverVer=09/01/2018,1.0.5.892
1&gt;C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(1634,5): warning : The driver package project C:\VS15\WinUSBPack\WinUSBPack\WinUSBPack.vcxproj doesn't have any references to other projects and no projects were packaged.
1&gt;C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(1634,5): warning :
1&gt;.......................
1&gt;Signability test complete.
1&gt;
1&gt;Errors:
1&gt;None
1&gt;
1&gt;Warnings:
1&gt;None
1&gt;
1&gt;Catalog generation complete.
1&gt;C:\VS15\WinUSBPack\x64\Debug\WinUSBPack\mydriver.cat
1&gt;SIGNTASK : SignTool error : No certificates were found that met all the given criteria.
1&gt;Done building project "WinUSBPack.vcxproj" -- FAILED.
1&gt;Done building project "WinUSBPack.vcxproj" -- FAILED.
========== Rebuild All: 0 succeeded, 1 failed, 0 skipped ==========

I'm writing a book to introduce WinUSB application for developers. Please fix it.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: dd1c106b-9499-4f15-f479-92537b5e4bed
* Version Independent ID: a10bfa35-ff3e-b169-0314-750dc5a28794
* Content: [WinUSB (Winusb.sys) Installation](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/winusb-installation#a-href-idhowtoahow-to-create-a-driver-package-that-installs-winusbsys)
* Content Source: [windows-driver-docs-pr/usbcon/winusb-installation.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/winusb-installation.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-08-31:
Hi @ahidaka, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @PageWriter-MSFT to this issue, and they'll triage this as soon as they have a moment.

ahidaka, 2018-08-31:
For your ref.

[image]
[image]

ahidaka, 2018-08-31:
Version Information.

Visual Studio Enterprise 2017 Version 15.8.2
Windows Driver Kit - Windows 10.0.17134.1

EliotSeattle, 2018-08-31:
Thanks, @ahidaka for the feedback and info. @PageWriter-MSFT can you please triage this to see if this is a doc issue, product issue, or both?

duncanmacmichael, 2019-08-08:
Realizing this is a year later, I'm coming to this now to try to determine if this is a product or documentation issue. On my own PC with Visual Studio 2019 and the WDK 1903, I created a new WinUSB driver package project and built it successfully with no issues. I don't see any immediate errors in the doc here, so I'm going to close this issue as it appears to have been fixed in the latest version of the WDK and Visual Studio.

closed 2019-08-08 by duncanmacmichael