MicrosoftDocs/windows-driver-docs #819: Causes and resolution steps

LuigiBruno, 2018-09-03:
Please provide a list of known causes as well as troubleshooting and resolution steps.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 05c732ce-fd6c-98e4-19ca-f52c6dd3e7b5
* Version Independent ID: 868b8c46-2096-9d10-2f6b-f015e6a57003
* Content: [Bug Check 0x154 UNEXPECTED_STORE_EXCEPTION](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x154--unexpected-store-exception)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x154--unexpected-store-exception.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x154--unexpected-store-exception.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

wdkbot, 2018-09-03:
Hi @LuigiBruno, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @DOMARS to this issue, and they'll triage this as soon as they have a moment.

DOMARS, 2018-09-06:
@LuigiBruno - I have started the work to expand this guidance as well as see if we can document arg3. If you have a dmp file handy, I would be curious what values you have seen for arg3 in !anaylyze.

DOMARS, 2018-09-07:
@LuigiBruno - I'm working with the dev team to expand this guidance.

LuigiBruno, 2018-09-29:
@DOMARS Thank you.

LuigiBruno, 2018-10-24:
@paulcbetts I think it is the Windows Store Component described at https://docs.microsoft.com/en-us/windows-hardware/manufacture/desktop/manage-the-component-store.

GernotB, 2019-02-20:
@LuigiBruno I think the Component Store is different from the store component. Component Store is for management of [side-by-side assemblies](https://docs.microsoft.com/en-us/dotnet/framework/app-domains/assemblies-and-side-by-side-execution) and I'd suppose the store component is a functional unit in the kernel. Otherwise it's hard to imagine it can throw a bug check.

alexanderhwhite, 2019-03-22:
@DOMARS, any update on known causes, troubleshooting, and resolution guidance yet?

chentiangemalc, 2019-05-24:
I’ve seen this a few times now on Windows 10 machines would love to see more guidance in this especially as I’m not getting any memory dump generated when it occurs.

LuigiBruno, 2019-06-26:
> @LuigiBruno I think the Component Store is different from the store component. Component Store is for management of side-by-side assemblies and I'd suppose the store component is a functional unit in the kernel. Otherwise it's hard to imagine it can throw a bug check.

Thanks @GernotB.

GiuseppeGiacoppo, 2019-08-07:
Hi everyone, I'm having this error for a while now and I cannot find any solution to this problem. Do you have any hint or advanced documentation on this critical event? Thanks

DOMARS, 2020-02-27:
@LuigiBruno and the others that have commented here - I added some additional information to indicate that this is a kernel memory component failure. I also added some types of things you can try if you encounter this type of error, and it is not occurring in driver code that you are debugging.  For additional guidance beyond the docs, I would recommend the kernel driver developer forums:

•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

Thanks for your interest in improving the stop code documentation.

closed 2020-02-27 by DOMARS

LuigiBruno, 2020-03-20:
> @LuigiBruno and the others that have commented here - I added some additional information to indicate that this is a kernel memory component failure. I also added some types of things you can try if [...]

Thank you @DOMARS.