MicrosoftDocs/windows-driver-docs #822: Missing entry for parameter 1 = 0x50565372

M4deman, 2018-09-05:
Parameter 1 can be 0x0000000050565372, but it is not listed in the table.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

wdkbot, 2018-09-05:
Hi @M4deman, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @DOMARS to this issue, and they'll triage this as soon as they have a moment.

DOMARS, 2018-09-14:
@M4deman - I'm not able to validate the memory parameter 1 for this stop code you describe:   0x50565372 
Could you double check that number and confirm this is with Windows 10?
If you have a dmp file handy and can paste in the output from (!analyze)[https://docs.microsoft.com/windows-hardware/drivers/debugger/-analyze] that would be helpful. Thanks for sharing your experience and your interest in improving our docs.

M4deman, 2018-09-16:
Hi, the code is correct but it was Windows 7. Sorry, I can't find any clue on this page, that it's only Windows 10 related.

DOMARS, 2018-09-17:
@M4deman - As mainstream support for Windows 7 ended in 2015, and we have limited resources to update the stop code documentation I won't be able to research and add information on the  0x50565372 parameter for  the 0X1A stop code. Good luck in your troubleshooting and thanks for the interest in our docs.

closed 2018-09-17 by DOMARS