MicrosoftDocs/windows-driver-docs #823: MBIM_MS_SET_NETWORK_BLACKLIST is not described anywhere.

janosvitok, 2018-09-06:
As subject says.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 199a05b2-82c8-ea4f-962a-f3c8e14d4703
* Version Independent ID: 6878502f-e58d-789d-5317-2f6b5332a548
* Content: [MB Network Blacklist Operations](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-network-blacklist-operations)
* Content Source: [windows-driver-docs-pr/network/mb-network-blacklist-operations.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/mb-network-blacklist-operations.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-09-06:
Hi @janosvitok, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @duncanmacmichael to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-09-06:
Thanks, @janosvitok. @duncanmacmichael, can you please set priority for this request for details about MBIM_MS_SET_NETWORK_BLACKLIST against your other deliverables? Thanks.

duncanmacmichael, 2018-09-10:
Hi @janosvitok , good catch! After double-checking the spec, it seems that MBIM_MS_NETWORK_BLACKLIST_INFO is not actually the correct structure. MBIM_MS_NETWORK_BLACKLIST_INFO is used in both the Set InformationBuffer and in responses/unsolicited events. I've corrected the page and the change should be live later today or tomorrow. Thanks for your interest in the docs!

closed 2018-09-10 by duncanmacmichael