MicrosoftDocs/windows-driver-docs #824: Numeric values of CIDs are missing

janosvitok, 2018-09-06:
MBIM_CID_BASE_STATIONS_INFO
MBIM_CID_LOCATION_INFO_STATUS

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 05ac7540-198a-9a9a-c8cd-1bb4eac81242
* Version Independent ID: 5223dc75-17e7-d9cd-94dd-d21704fa331c
* Content: [MB base stations information query support](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-base-stations-information-query-support)
* Content Source: [windows-driver-docs-pr/network/mb-base-stations-information-query-support.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/mb-base-stations-information-query-support.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-09-06:
Hi @janosvitok, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @duncanmacmichael to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-09-06:
Thanks, @janosvitok 
@duncanmacmichael can you please look into this?

duncanmacmichael, 2018-09-10:
Hi @janosvitok , looks like the command codes for these CIDs are indeed missing. I'll see if I can track them down for you and will update the page when I have the correct answer from the feature team.

duncanmacmichael, 2018-09-11:
Hi @janosvitok , I've gotten a response from the feature team for the correct numeric values. I added them to the page, and the changes should be live tomorrow. One thing to note is that the MBIM_CID_LOCATION_INFO_STATUS CID is _defined_ starting in Windows 10, version 1709, but is not currently supported/handled by the OS. 

Thanks again for your interest in the docs!

closed 2018-09-11 by duncanmacmichael