MicrosoftDocs/windows-driver-docs #827: How does this constraint work?

CKroid, 2018-09-07:
The first sentence mentioned "Starting in Windows 10 version 1709, you can specify that a Universal Windows Platform (UWP) app should only load if a specific driver is present."
In a scenario where a new UWP app requires a new version of driver, but the system is having the old driver, this new UWP app will still be available though Windows Store, and after the user updated to this new UWP app, the user will no longer be able to run the app anymore until he updates the driver?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d63fca17-2476-874f-5329-5f065552ddad
* Version Independent ID: c5470d95-0ead-9d43-1e6c-04af1acd79bc
* Content: [Pairing a driver with a Universal Windows Platform (UWP) app](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/pairing-app-and-driver-versions)
* Content Source: [windows-driver-docs-pr/install/pairing-app-and-driver-versions.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/pairing-app-and-driver-versions.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-09-07:
Hi @CKroid, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @tedhudek to this issue, and they'll triage this as soon as they have a moment.

tedhudek, 2018-09-07:
Hi @CKroid thanks for the feedback. I have added a sentence to https://docs.microsoft.com/windows-hardware/drivers/install/pairing-app-and-driver-versions to clarify this behavior.  Please see the second sentence on that page.

closed 2018-09-07 by tedhudek

tedhudek, 2018-09-10:
Hi @CKroid I'd recommend leaving up all versions of the app that anyone might be using with older driver versions.