MicrosoftDocs/windows-driver-docs #829: Suggestion: convert table to paragraphs with heading (and anchors)

kITerE, 2018-09-08:
This will make the link from outside to a specific macro.
Example:

## PsGetCurrentProcess &lt;a name="psgetcurrentprocess"&gt;&lt;/a&gt;

Declared in: Ntddk.h
Returns a pointer to the process of the current thread.
&lt;…&gt;

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ed5ff8a0-637e-6e3d-4b7e-652c1934ecdc
* Version Independent ID: 6bdcbd7f-3142-1646-2c03-dd31cf33f58f
* Content: [Windows kernel macros](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/mm-bad-pointer)
* Content Source: [windows-driver-docs-pr/kernel/mm-bad-pointer.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/mm-bad-pointer.md)
* GitHub Login: @windows-driver-content

wdkbot, 2018-09-08:
Hi @kITerE, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @PageWriter-MSFT to this issue, and they'll triage this as soon as they have a moment.

tedhudek, 2018-12-12:
Hi @kITerE sorry for the delay on this one. I think this is a fabulous idea. Do you have any interest in submitting a PR with your proposed change?

tedhudek, 2018-12-20:
Update -- I just pushed the change to the original page live.  Still plan to update links to point back to the various sections.

https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/mm-bad-pointer

kITerE, 2018-12-22:
Hi @tedhudek, thank you

closed 2018-12-22 by kITerE