MicrosoftDocs/windows-driver-docs #831: Inconsistent UUIDs defined for Basic Connectivity Extensions service

cbchan, 2018-09-11:
According to different Microsoft documentation (see the list below), Microsoft has defined inconsistent UUIDs for the same Basic Connectivity Extensions service. Two variants are observed so far, where they differ only by a single nibble: 
  3d01dcc5-fef5-4d05-[ 9 ]d3a-bef7058e9aaf
  3d01dcc5-fef5-4d05-[ 0 ]d3a-bef7058e9aaf

As neither UUID is listed on http://compliance.usb.org/mbim/, it's hard to confirm which one is the correct, or if they have been incorrectly used interchangeably on the Windows side.

References:
The following CIDs use UUID 3d01dcc5-fef5-4d05-9d3a-bef7058e9aaf
https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-modem-reset-operations
https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-protocol-configuration-operations--pco-

The following CIDs use UUID 3d01dcc5-fef5-4d05-0d3a-bef7058e9aaf
https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-multi-sim-operations
https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-network-blacklist-operations
https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-base-stations-information-query-support
https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-provisioned-context-operations
https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-lte-attach-operations:

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c6d93fd2-1e34-19a2-c0cb-598723199c4c
* Version Independent ID: 22ec6503-7f14-1319-db82-cda1ccab0754
* Content: [MB Protocol Configuration Operations (PCO)](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-protocol-configuration-operations--pco-)
* Content Source: [windows-driver-docs-pr/network/mb-protocol-configuration-operations--pco-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/mb-protocol-configuration-operations--pco-.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-09-11:
Hi @cbchan, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @duncanmacmichael to this issue, and they'll triage this as soon as they have a moment.

aleksander0m, 2018-09-11:
For reference, the Qualcomm-based Dell/Foxconn DW5821e uses 0d3a, while the Intel based Fibocom L850-GL uses 9d3a.

duncanmacmichael, 2018-09-11:
Hi @cbchan , thanks for the question. I'm not sure about the exact answer, so I'll ask the mobile broadband feature team what we should say here.

duncanmacmichael, 2018-09-11:
Hi @cbchan , I've gotten a response from the feature team for you. The UUID with a 9 in that nibble is in fact incorrect, as the UUID for the Basic Connectivity Extensions service is 3d01dcc5-fef5-4d05-0d3a-bef7058e9aaf. I'll correct the two pages with the typo on them, and the changes should be live later today. Thanks for helping to improve the docs!

closed 2018-09-11 by duncanmacmichael