MicrosoftDocs/windows-driver-docs #832: TraceView BUG

kouzhudong, 2018-09-11:
在Windows 2003 的64位操作系统上，用TraceView打开PDB文件，TraceView闪退。

TraceView可以是任何版本的。
PDB可以是任何编译器生成的。

https://correy.webs.com.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ce0e2d68-b7a9-9bcd-92c1-0e7c40ed6b72
* Version Independent ID: ad275cd3-f168-ba9d-385d-b1a9e69df296
* Content: [TraceView](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/traceview#)
* Content Source: [windows-driver-docs-pr/devtest/traceview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/traceview.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-09-11:
Hi @kouzhudong, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @EliotSeattle to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-09-12:
Hi @kouzhudong. I don't understand what your issue is with the documentation. Can you  explain a little more?

kouzhudong, 2018-09-13:
in windows server 2003, if traceview.exe open an pdb file, it must be quickly exit.

EliotSeattle, 2018-09-13:
Hi @kouzhudong.
Are you asking that we add a comment to that effect? I'm so sorry, but I am still not clear. Something like, "If you use traceview.exe to open a pdb file running on Windows Server 2003, then you must _____________." Are you saying the tester must exit traceview quickly, close the pdb file quickly, or something else? What do you mean by quickly? And what is the behavior if you don't exit?

Sorry if I am not getting your meaning.

kouzhudong, 2018-09-14:
in windows server 2003, if traceview.exe open an pdb file, it automatic exits.

微软的一群低级弱智智能儿。

EliotSeattle, 2018-09-14:
Thanks for being patient with me, @kouzhudong. I added a note in [Creating a Trace Session with a PDB File](https://docs.microsoft.com/windows-hardware/drivers/devtest/creating-a-trace-session-with-a-pdb-file):  "If you use TraceView to open a PDB file on a computer running Windows Server 2003, TraceView automatically exits." It should be live later today.

closed 2018-09-14 by EliotSeattle

kouzhudong, 2018-09-15:
This is the end.
Not modified.
Doesn't support 2003?

EliotSeattle, 2018-09-17:
I am afraid not, @kouzhudong. Support for Windows Server 2003 ended in 2015. See https://www.microsoft.com/en-us/cloud-platform/windows-server-2003.