MicrosoftDocs/windows-driver-docs #833: Should this be called Protocol Configuration Options?

janosvitok, 2018-09-11:
TS 24.008 does not mention any Protocol Configuration Operations, but mentions Protocol Configuration Options pretty often.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c6d93fd2-1e34-19a2-c0cb-598723199c4c
* Version Independent ID: 22ec6503-7f14-1319-db82-cda1ccab0754
* Content: [MB Protocol Configuration Operations (PCO)](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-protocol-configuration-operations--pco-#modem-behavior-with-selective-suspend-and-connected-standby)
* Content Source: [windows-driver-docs-pr/network/mb-protocol-configuration-operations--pco-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/mb-protocol-configuration-operations--pco-.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-09-11:
Hi @janosvitok, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @duncanmacmichael to this issue, and they'll triage this as soon as they have a moment.

janosvitok, 2018-09-11:
Perhaps only the first sentence should be changed:
> Traditionally, the Windows NDIS definitions for Protocol Configuration **Operations** (PCO) values have been generic, for potentially receiving full PCO values from the modem and network in the future

duncanmacmichael, 2018-09-11:
Hi @janosvitok , good catch. Looks like you're correct; PCO does stand for Protocol Configuration Options, but the title is about operations with PCO so I probably switched those words in my head. I'll change the title to "MB Protocol Configuration Options (PCO) operations," which is more accurate. Thanks for helping to clarify this! The changes should be live later today.

closed 2018-09-11 by duncanmacmichael