MicrosoftDocs/windows-driver-docs #834: Need more links

jakeoshins, 2018-09-11:
In sentences like "The given value-entry-name and/or value is of the registry type REG_EXPAND_SZ." there should be a link to the definition of REG_EXPAND_SZ.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: acdc91cd-c89e-684c-c48b-5d66d098fa9c
* Version Independent ID: 4865d5a4-df22-1c26-f993-a661882194a7
* Content: [INF AddReg Directive](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-addreg-directive)
* Content Source: [windows-driver-docs-pr/install/inf-addreg-directive.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-addreg-directive.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-09-11:
Hi @jakeoshins, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @tedhudek to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-09-11:
Thanks, @jakeoshins. I added links to https://docs.microsoft.com/en-us/windows/desktop/SysInfo/registry-value-types for info about the unexplained types.

closed 2018-09-11 by EliotSeattle

tedhudek, 2018-09-11:
@jakeoshins do you think it would be useful to update all refs to REG_* to link to https://docs.microsoft.com/en-us/windows/desktop/SysInfo/registry-value-types?

tedhudek, 2018-09-14:
Hi @jakeoshins I added some links per your request. There are actually a bunch of edge cases involved doing this programmatically across two repos that I don't have time to comb through at the moment. The ones I added will be live in a few minutes.  Thx for the suggestion!