MicrosoftDocs/windows-driver-docs #835: Broken links

englishb, 2018-09-11:
All six of the articles about Multiplane Overlay Support linked to from this page are broken links that redirect to the top-level page "Programming reference for Windows Driver Kit (WDK)" at https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/index
I cannot find these articles through MSDN search, bing, google, or anything else.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5fc77c70-4150-73f3-f826-d436b901413d
* Version Independent ID: 226779ce-ad8c-6195-89b6-a8b63dfdafc8
* Content: [Multiplane overlay support](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/multiplane-overlay-support)
* Content Source: [windows-driver-docs-pr/display/multiplane-overlay-support.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/multiplane-overlay-support.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-09-11:
Hi @englishb, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @mimisasouvanh to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-09-11:
Thanks, @englishb 
@mimisasouvanh these look to be OVW pages in WDCML that did not get migrated. Can you please take a look? Thanks!

mimisasouvanh, 2018-09-25:
Hi @englishb 
I have inserted links to relevant DDIs for this topic. They'll be published in the next 24 hours.
Thanks,
Mimi

closed 2018-09-25 by mimisasouvanh