MicrosoftDocs/windows-driver-docs #836: on access scanner need

AkshayUsha, 2018-09-12:
on access scanner need microsoft can help me pls....

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d6eece4e-03de-80b6-dba7-e475c0f0e575
* Version Independent ID: 682f3431-07d6-b9c1-5b2d-71dfe92859e7
* Content: [File System Minifilter Drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/file-system-minifilter-drivers)
* Content Source: [windows-driver-docs-pr/ifs/file-system-minifilter-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/file-system-minifilter-drivers.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

wdkbot, 2018-09-12:
Hi @AkshayUsha, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @EliotSeattle to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2018-09-12:
Hi @AkshayUsha. Are you asking about an antivirus scanner? I'm unclear about the issue you are reporting.

AkshayUsha, 2018-09-13:
yes i m asking about antivirus scanner

AkshayUsha, 2018-09-13:
> Hi @AkshayUsha. Are you asking about an antivirus scanner? I'm unclear about the issue you are reporting.

yes

EliotSeattle, 2018-09-13:
Thank you, @AkshayUsha. What exactly are you asking about? Is this about the documentation in this section? 
Or is this a question about using an access scanner? If about usage, the best place to get a response would be in the [WDK forum](https://social.msdn.microsoft.com/Forums/windowshardware/en-US/home?forum=wdk). If it's about the documentation, I can help.

kITerE, 2018-09-28:
@AkshayUsha, https://github.com/Microsoft/Windows-driver-samples/tree/master/filesys/miniFilter/avscan

EliotSeattle, 2018-10-06:
@barrygolden can you look into this question regarding the sample?

EliotSeattle, 2019-03-06:
Closing because of inactivity.

closed 2019-03-06 by EliotSeattle