MicrosoftDocs/windows-driver-docs #839: Please update/correct the Commit Status values list

NileshMali1, 2018-09-13:
Hi There,

The documentation is not updated and misleading at some places, which causes a lot jumping here and there in the doc pages and try outs while coding to get to correct information. I would highly appreciate if the doc gets revised and updated to reflect correct information.

Also, when do you get the Commit Received state? I understand it must be right after Commit Pending state once user commits; however, when user commits the API returns response code 202 (accepted) with empty body and then Commit Status gets updated to either Commit Complete state or Commit Failed state, there is no use of Commit Received state unless it is returned in the response body when user commits. Could you please have a look  and make necessary updates?

Best regards,
NM

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0ec4d9d1-95d0-aaf4-5bc2-1fd0ff3d6ed0
* Version Independent ID: a506d854-e239-58d1-342c-99622ee5249b
* Content: [| Microsoft Docs](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/manage-product-submissions)
* Content Source: [windows-driver-docs-pr/dashboard/manage-product-submissions.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/manage-product-submissions.md)
* Product: **windows-hardware**
* GitHub Login: @balapv
* Microsoft Alias: **balapv**

wdkbot, 2018-09-13:
Hi @NileshMali1, I'm the Windows Driver Docs assign bot. Thanks for your interest in the docs! I'm assigning @EliotSeattle to this issue, and they'll triage this as soon as they have a moment.

EliotSeattle, 2019-03-06:
@NileshMali1 Sorry for the delay on this.
@balapv can you answer these questions?

GabeSJones, 2019-03-12:
I have a similar set of questions.  The current page shows "commitReceived," "commitCompleted," and "commitFailed" as possible values of "commitStatus."  The page at https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/commit-a-product-submission only lists "commitStarted" as a possible value.  The page at https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-product-data#submission-resource only lists "commitPending" as a possible value.  I would like to see these pages agree with each other and list all of the possible values and when they are expected to occur.

tedhudek, 2020-11-09:
Thanks for reaching out - sorry it took us a while.  Please see related doc updates in 7f72cade9.

closed 2020-11-09 by tedhudek