MicrosoftDocs/windows-driver-docs #845: chkINF not found

rena2019, 2018-09-18:
there is no Tools\chkinf subdirectory in the current WDK anymore

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 311fd11a-8230-cd6b-1c52-761cfc56e86b
* Version Independent ID: f9e6d391-3a91-a24f-cc1b-540216d29c66
* Content: [ChkINF](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/chkinf)
* Content Source: [windows-driver-docs-pr/devtest/chkinf.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/chkinf.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-09-18:
Thanks. Looking into this.

EliotSeattle, 2018-09-19:
Right you are, @rena2019. ChkINF no longer ships with the WDK. I will take down the documentation. Thank you so much for pointing this out!

tedhudek, 2018-09-21:
Thanks @rena2019 you are correct, InfVerif has replaced the old ChkINF tool.  I've redirected the ChkINF doc pages to go to https://docs.microsoft.com/windows-hardware/drivers/devtest/infverif.

closed 2018-09-21 by tedhudek