MicrosoftDocs/windows-driver-docs #846: DRIVER_PNP_WATCHDOG information missing

LuigiBruno, 2018-09-18:
Add a page to describe the DRIVER_PNP_WATCHDOG bug check.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ddc632b2-f4e1-b784-dd1e-878feac077f9
* Version Independent ID: 7a6866f4-3333-f2d1-86d7-dbb3d00925e0
* Content: [Bug Check Code Reference](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-code-reference2)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-code-reference2.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-code-reference2.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

LuigiBruno, 2018-09-29:
@DOMARS - Yes, it looks like: I've never seen DRIVER_PNP_WATCHDOG bug check before. Thank you.

DOMARS, 2019-09-11:
@LuigiBruno - We published the DRIVER_PNP_WATCHDOG topic, so closing this issue.

https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1d5--driver-pnp-watchdog

closed 2019-09-11 by DOMARS