MicrosoftDocs/windows-driver-docs #848: Typo in windbg command?

jjbengo, 2018-09-19:
The backslash in commands "x notepad!\*" and "x notepad!*main\*" makes them produce no output. I think the proper commands should not contain "\": "x notepad!*" and "x notepad!*main*"

---
#### Detalles del documento

⚠ *No edite esta sección. Se requiere para docs.microsoft.com ➟ Vinculación de problema de GitHub.*

* ID: 142adea7-b7b8-4a5a-5e19-83a202ab04b2
* Version Independent ID: 489c0c22-908c-aa93-520c-836c941b5b4e
* Content: [Getting Started with WinDbg (User-Mode)](https://docs.microsoft.com/es-es/windows-hardware/drivers/debugger/getting-started-with-windbg)
* Content Source: [windows-driver-docs-pr/debugger/getting-started-with-windbg.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/getting-started-with-windbg.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-09-19:
@jjbengo - I will take a look and see. This may take a bit as we are working on docs for the rs5 release in the next few weeks, but thanks for haring your experience with this lab.

DOMARS, 2018-10-10:
@jjbengo - The rendering issues are fixed. Thanks again for reporting this issue and good luck with your debugging.

closed 2018-10-10 by DOMARS