MicrosoftDocs/windows-driver-docs #849: Need Info Between {}

GGabc, 2018-09-19:
Nice article, but it lacks how to determine the GUID for use between the {} on the BCDedit commandline. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1b8e80b1-789d-942c-d2ea-f7b3af8df312
* Version Independent ID: b93ffbfc-b7a2-5f23-c5aa-0b56eea5b7a9
* Content: [Boot Parameters to Manipulate Memory](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/boot-parameters-to-manipulate-memory)
* Content Source: [windows-driver-docs-pr/devtest/boot-parameters-to-manipulate-memory.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/boot-parameters-to-manipulate-memory.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-09-20:
Hi @GGabc.
Did you take a look at [BCDEdit /set](https://docs.microsoft.com/windows-hardware/drivers/devtest/bcdedit--set)? Is this the kind of thing you're looking for:

Parameters
[{ID}]
The {ID} is the GUID that is associated with the boot entry. If you do not specify an {ID}, the command modifies the current operating system boot entry. If a boot entry is specified, the GUID associated with the boot entry must be enclosed in braces { }. **To view the GUID identifiers for all of the active boot entries, use the bcdedit /enum command**. The identifier for the current boot entry is {current}. For more information about this option, use the following command: bcdedit /? ID. (my emphasis)

EliotSeattle, 2019-03-06:
Closing without any reply back.

closed 2019-03-06 by EliotSeattle