MicrosoftDocs/windows-driver-docs #851: Broken links - Get SDK/WDK Insider Preview

bityob, 2018-09-22:
Both two links leads to a not found page. 

Get SDK Insider Preview
https://www.microsoft.com/software-download/windowsinsiderpreviewWDK

Get WDK Insider Preview
https://www.microsoft.com/software-download/windowsinsiderpreviewWDK

Yes, this is the the same link.

Thanks.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 34d06a09-bca8-3177-00fd-2195f56c4fb9
* Version Independent ID: 5328b553-ad08-4808-0b14-9720484040ce
* Content: [Installing preview versions of the Windows Driver Kit (WDK)](https://docs.microsoft.com/en-us/windows-hardware/drivers/installing-preview-versions-wdk)
* Content Source: [windows-driver-docs-pr/installing-preview-versions-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/installing-preview-versions-wdk.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

closed 2018-10-06 by EliotSeattle

BerkanYildiz, 2018-12-29:
The links are invalid for me too.

PeterPann23, 2019-08-25:
Hi @EliotSeattle , your link seems to be broken as well