MicrosoftDocs/windows-driver-docs #860: Global Admin credentials

gonzalre, 2018-09-27:
Please note that the term Global Admin credentials it is no about Azure AD Global Admin, but to the developer portal Global Admin account, that at the end, are to different things. Please address this as this could mislead our customers.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1b50e13d-f920-d392-6be0-c792bceac73c
* Version Independent ID: 2c10bcf7-5a2a-26c9-af2d-bdae6845159f
* Content: [Adding or Removing Users](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/adding-or-removing-users)
* Content Source: [windows-driver-docs-pr/dashboard/adding-or-removing-users.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/adding-or-removing-users.md)
* Product: **windows-hardware**
* GitHub Login: @aahill
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-09-27:
Thanks, @gonzalre. I've edited the opening of the topic to differentiate the two account credentials. Should be live later today.

closed 2018-09-27 by EliotSeattle

gonzalre, 2018-09-27:
Great! Thanks!

Gonzalo Reyna (GONZ)
Support Escalation Engineer
Microsoft Azure Rapid Response
Office: +1 (469)-775-2554
gonzalre@microsoft.com<mailto:gonzalre@microsoft.com>
M-F 8:00-17:00 CST
[cid:image001.jpg@01D45420.D7BAD7A0]

If you have any feedback about my work, please let either me or my manager Harry Storey know at harryst@microsoft.com<mailto:harryst@microsoft.com>