MicrosoftDocs/windows-driver-docs #861: HTML <title>contains extra escaping

kITerE, 2018-09-27:
```
	&lt;title&gt;FLT\_PARAMETERS for IRP\_MJ\_ACQUIRE\_FOR\_SECTION\_SYNCHRONIZATION union | Microsoft Docs&lt;/title&gt;
```

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 20a64a62-c8b8-d3f7-942a-145b8d8907c9
* Version Independent ID: 3ecdda0f-ef9b-5bde-431e-81a7af013f64
* Content: [FLT\_PARAMETERS for IRP\_MJ\_ACQUIRE\_FOR\_SECTION\_SYNCHRONIZATION union](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/flt-parameters-for-irp-mj-acquire-for-section-synchronization)
* Content Source: [windows-driver-docs-pr/ifs/flt-parameters-for-irp-mj-acquire-for-section-synchronization.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/flt-parameters-for-irp-mj-acquire-for-section-synchronization.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

kITerE, 2018-09-27:
PR #862 (fix)

tedhudek, 2018-09-27:
PR is merged.  Thanks!

closed 2018-09-27 by tedhudek