MicrosoftDocs/windows-driver-docs #872: Drivers for Windows

johnsmith2679, 2018-10-05:
I want to update my drivers of Windows 7 I need 64-bit supportable drivers. My older driver is not compatible with Microsoft Office soI want to replace this newer one. IS there any way to replace the driver package without uninstalling or deleting the other software.  Anyone have a query related to their Microsoft Office and other Microsoft plugins like installation guide, not responding, slow speed of office etc follow  https://www.msofficetechnicalsupportnumbers.com/ here you can fix all error of MS Office.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6e65610e-d587-acd5-6173-d304e5224daa
* Version Independent ID: 2db80595-55a2-bb08-d1af-4888ea8c33c5
* Content: [Mapping Driver Files](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/mapping-driver-files)
* Content Source: [windows-driver-docs-pr/debugger/mapping-driver-files.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/mapping-driver-files.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-10-09:
Thanks for reaching out to the Windows Driver Developer Documentation team. Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation. Best of luck with this issue.

closed 2018-10-09 by DOMARS

rodwiddowson, 2018-10-09:
As an outside I'd add that you might want to pop in to the [NTDEV](https://community.osr.com/categories/ntdev) list