MicrosoftDocs/windows-driver-docs #874: Links to EDID structures broken

BrianCatlin, 2018-10-05:
The two links to the EDID structure don't work

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 87e21e48-58df-e60c-e125-644a41e7fb6f
* Version Independent ID: 13071a05-b0d4-4e93-0ddd-e9b7ddc492b7
* Content: [Monitor Class Function Driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/monitor-class-function-driver)
* Content Source: [windows-driver-docs-pr/display/monitor-class-function-driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/monitor-class-function-driver.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-10-05:
Hi @BrianCatlin.
I've updated the topic. The links were to the glossary that we retired, so I incorporated the text from the glossary into the topic. I suspect that it might be less that you're looking for, though, as it tell what an EDID is and how to find one, but does not offer instructive advice on using an EDID. I'll leave this issue open and let @mimisasouvanh think about next steps. Here's the text. It's PR is in the queue and should be live shortly.

"A monitor stores its identification and capability information in an Extended Display Identification Data (EDID) structure, a format that lets the display supply the host with information about its identity and capabilities independent of the communications protocol used between the monitor and host. A request, from a user-mode application, to read a monitor's EDID is processed by the function driver (Monitor.sys) in that monitor's device stack. When the monitor function driver receives a request to retrieve the monitor's EDID, it sends a request to the display port/miniport driver pair that is represented by the physical device object (PDO) at the bottom of the monitor's device stack. The display port/miniport driver pair uses the Display Data Channel (DDC) protocol to read the monitor's EDID over the I²C bus, which is a simple two-wire bus built into all standard monitor cables.

The EDID can be obtained using the ACPI_METHOD_OUTPUT_DDC method whose alias is defined in Dispmprt.h. This method is required for integrated LCDs that do not have another standard mechanism for returning EDID data."

mimisasouvanh, 2018-10-09:
I've added links to the mentioned structures.

closed 2018-10-09 by mimisasouvanh