MicrosoftDocs/windows-driver-docs #875: Content useless

BrianCatlin, 2018-10-05:
Monitor filter drivers are supported, but where is the description of how the monitor class driver works? Where is the list of IRPs that the class function driver supports? Without either of those, there is no way to write a monitor filter driver! Most of the links on this page are broken. Is there a sample monitor filter driver in the WDK?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 87e21e48-58df-e60c-e125-644a41e7fb6f
* Version Independent ID: 13071a05-b0d4-4e93-0ddd-e9b7ddc492b7
* Content: [Monitor Class Function Driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/monitor-class-function-driver)
* Content Source: [windows-driver-docs-pr/display/monitor-class-function-driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/monitor-class-function-driver.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

mimisasouvanh, 2018-10-09:
Thank you for your feedback! I went ahead and added the links that were missing to this topic, including a link to I2C Bus and Child Devices of the Display Adapter, which contains the sections pointing out I2C functions, and I2C functions implemented by the Video Port Driver. I am currently investigating the list of IRPs, and will update it if I get that info. I have merged this issue with #876 and will now close this issue.

closed 2018-10-09 by mimisasouvanh

BrianCatlin, 2018-10-09:
That is a start, but there still needs to be a theory of operation section that describes what monitor filter drivers can and should do