MicrosoftDocs/windows-driver-docs #876: Useless!

BrianCatlin, 2018-10-05:
So, how EXACTLY does someone write a monitor filter driver? There is nothing here that provides any information other than that monitor filter drivers are supported

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ce19b390-c50f-5a79-a483-ff5de162ac91
* Version Independent ID: 5d0265b5-9492-ace8-84c9-466ecbc48c9a
* Content: [Monitor Filter Drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/monitor-filter-drivers)
* Content Source: [windows-driver-docs-pr/display/monitor-filter-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/monitor-filter-drivers.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

mimisasouvanh, 2018-10-09:
This issue is the same request as issue #875. Closing #875 .

mimisasouvanh, 2018-10-29:
Hi @BrianCatlin, I've created a task to improve this topic.

closed 2018-10-29 by mimisasouvanh