MicrosoftDocs/windows-driver-docs #878: Download WDK for Windows 10, version 1809 link is broken

spayne, 2018-10-05:
The link to the wdk: Download WDK for Windows 10, version 1809 https://go.microsoft.com/fwlink/?linkid=2026156 is broken for me.  

I expect that clicking on it take me to a download, but instead it takes me to a generic/top levlel Microsoft site (https://www.microsoft.com/en-us/).  

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5930505f-6810-b2a2-86b5-3c678aae7f89
* Version Independent ID: 28315cf8-f06f-23ad-2088-9d51d08efe0d
* Content: [Download the Windows Driver Kit (WDK)](https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk)
* Content Source: [windows-driver-docs-pr/download-the-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-10-05:
Holy cow, thanks, @spayne 
Not sure what's going on with our tooling. Am looking into it. In the meantime, the direct link is [http://download.microsoft.com/download/1/4/0/140EBDB7-F631-4191-9DC0-31C8ECB8A11F/wdk/wdksetup.exe](http://download.microsoft.com/download/1/4/0/140EBDB7-F631-4191-9DC0-31C8ECB8A11F/wdk/wdksetup.exe). Let me know if that doesn't work for you.

EliotSeattle, 2018-10-05:
Back to normal. Thanks for the heads up!

closed 2018-10-05 by EliotSeattle