MicrosoftDocs/windows-driver-docs #893: Add further information

LuigiBruno, 2018-10-10:
Add information about the meaning of each parameter, possible causes and resolution steps.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: aec3b612-e03e-f2d5-62de-e2edcbf5377f
* Version Independent ID: d4465366-fd3e-a92a-3b12-d7936ff4c049
* Content: [Bug Check 0x9 IRQL_NOT_GREATER_OR_EQUAL](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x9--irql-not-greater-or-equal)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x9--irql-not-greater-or-equal.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x9--irql-not-greater-or-equal.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-10-25:
First off, thanks for all of the input on the bug checks. We use customer input like this as well as data on how often the bug checks occur, to determine how we allocate our documentation team resources. 

I wanted to provide some additional information about how I prioritize the stop code content. My current focus is to document—for developers—the new stop codes such as DRIVER_PNP_WATCHDOG  that you mentioned in this issue: https://github.com/MicrosoftDocs/windows-driver-docs/issues/846. I also want to expand the general guidance we provide on dump files and stop codes in topics, like this one: https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/analyzing-a-kernel-mode-dump-file-with-kd

It takes a lot of time to research, write up, and collaborate with the dev team to create or update bug check topics. Sadly, there are more topics in need of more info than there are hours that I work. We have found that about 25 of the stop codes generate the vast majority of the customer issues, so that is where I focus our limited resources. 

For the existing stop codes, let me know if you could point out a handful of specific codes where additional information would be particularly helpful. 

Our goal is to document for a wide variety of situations why a specific bug check can occur. In the past, I’ve had to clean out some references to “this bug check is caused by CDROM drives” or “when a floppy disk does this,” as these describe specific technologies, at a specific point in time, rather than the underlying code error that causes the bug check. 

One additional caveat: We can’t disclose any data about the internal operations of Windows that wouldn’t be appropriate.  

As you might imagine, we are focused on Windows 10, as we think that is the most secure and reliable version of Windows, that we hope everyone upgrades to as soon as possible.

Thanks for your contributions and interest in improving our docs.

DOMARS, 2019-10-08:
@LuigiBruno - I have opened an internal work item to track this work, so closing this issue for now.

As I have mentioned before, we use a process to prioritize the most frequent bug checks out of the hundreds that are currently defined. With our limited documentation team resources, it is likely that we will only be able to do this work for a small sub set of all of the possible stop codes.

Thank you for your interest in improving our stop code docs.

closed 2019-10-08 by DOMARS

LuigiBruno, 2019-10-09:
> @LuigiBruno - I have opened an internal work item to track this work, so closing this issue for now.
> As I have mentioned before, we use a process to prioritize the most frequent bug checks out of [...]

OK @DOMARS. Thank you.