MicrosoftDocs/windows-driver-docs #918: Causes and resolutions steps needed

LuigiBruno, 2018-10-12:
Add information about possible causes and resolutions steps for this bug check.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c1307f8e-62f1-03c0-b953-2033d1fc9b3c
* Version Independent ID: 24dd7983-4656-9a66-e0cd-94dfd54be4e0
* Content: [Bug Check 0x14F PDC_WATCHDOG_TIMEOUT](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x14f--pdc-watchdog-timeout)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x14f--pdc-watchdog-timeout.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x14f--pdc-watchdog-timeout.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

EliotSeattle, 2018-10-12:
Thanks, @LuigiBruno. We'll add this to your other requests. :-)

Reincarnationist, 2019-07-12:
bluescree appears after setting max process number to 128 then run a bash fork bomb...

DOMARS, 2019-09-11:
@LuigiBruno - I have opened an internal work item to track this work, so closing this issue for now.

As I have mentioned before, we use a process to prioritize the most frequent bug checks out of the hundreds that are currently defined. With our limited documentation team resources, it is likely that we will only be able to do this work for a small sub set of all of the possible stop codes.

Thank you for your interest in improving our stop code docs.

closed 2019-09-11 by DOMARS