MicrosoftDocs/windows-driver-docs #921: Values for parameter 1

LuigiBruno, 2018-10-12:
Provide a list of known violations (values for parameter 1) and their corresponding meanings.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 11e0dfb6-99ee-e54c-0f47-111db569266a
* Version Independent ID: c26569e3-3d91-f15c-906d-0434c2fa8bdc
* Content: [Bug Check 0x121 DRIVER_VIOLATION](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x121---driver-violation)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x121---driver-violation.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x121---driver-violation.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-01-18:
@LuigiBruno - Thanks for this issue. Because we have not documented information on this bug check before, I would need to check with the dev team to be able to do that. As I have mentioned before, we use a process to prioritize the most frequent bug checks out of the hundreds that are currently defined. With our limited documentation team resources, it is likely that we will only be able to do this work for a small sub set of all of the possible stop codes.
Right now I'm working on the new stop codes that we have not documented at all.
Thank you for your interest in improving our stop code docs.

DOMARS, 2020-02-13:
@LuigiBruno - Added the description of param1, so closing this issue. Thanks for filing this issue and your interest in improving the stop code docs.

closed 2020-02-13 by DOMARS