MicrosoftDocs/windows-driver-docs #947: How to make WinDbg Watch value lines to wrap

agvardha, 2018-10-16:
I am having JSON strings in the WinDbg watch window and they go to infinity. How do I make the Watch Windows Wrap or when we use JAVA there is a neat way of seeing the entire JSON as a file ... Command window does wrap but watch and local window do not wrap!! Any tricks? 
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2d04112f-51f7-5886-5966-5515b1476c84
* Version Independent ID: f30e8014-3386-c08b-2644-6d32eff5f823
* Content: [Using the Watch Window](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/watch-window)
* Content Source: [windows-driver-docs-pr/debugger/watch-window.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/watch-window.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

agvardha, 2018-10-18:
@DOMARS any news? Displaying JSON string is so critical and now I have to drag the local/watch window across 4 monitors to see even then it becomes difficult. I am 100% sure such a fundamental fix MSFT could not have missed. It is just **my ignorance** so please help me out ...

DOMARS, 2018-10-25:
@agvardha - Thanks for reaching out to the Windows Driver Developer Documentation team. Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation.
You may have success posting your question in the following forum, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.
- [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

You may want to explore if it would be possible to use a debugger command to display the values that you are interested in.

closed 2018-10-25 by DOMARS