MicrosoftDocs/windows-driver-docs #957: No longer supported?

Wumpf, 2018-10-20:
&gt; This feature is not available in Windows 10, version 1507 and later versions of the WDK.
Why exactly? Is there a workaround or am I forced to set up a second physical computer?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 49640a8f-da9e-9977-3bb9-82dd29c020af
* Version Independent ID: e794cce0-71cc-f58b-7ece-0c9c76943f45
* Content: [Setting Up Kernel-Mode Debugging of a Virtual Machine in Visual Studio](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/setting-up-a-connection-to-a-virtual-machine-in-visual-studio)
* Content Source: [windows-driver-docs-pr/debugger/setting-up-a-connection-to-a-virtual-machine-in-visual-studio.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/setting-up-a-connection-to-a-virtual-machine-in-visual-studio.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

Wumpf, 2018-10-20:
Since it says this on all Visual Studio driver development pages on debugging (even on UMDF), does that mean that any driver debugging support from Visual Studio was dropped?

DOMARS, 2018-10-24:
@Wumpf - Thanks for reaching out to the Windows Driver Developer Documentation team. Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation.
You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.
•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
•	[OSR's ntdev List](https://www.osronline.com/showlists.cfm?list=ntdev)

Many driver developer find that because new driver code can destabilize the PC, using two PCs, is the best approach for commercial driver development.
 
To provide product input on the Visual Studio debugging support follow this link to file product feedback using the Windows Feedback Hub: 
http://aka.ms/dexex 
The Feedback Hub allows customers to file and up vote product issues. The link automatically sets the product category to Debugging Tools for Windows.
Thanks again for reaching out to the Windows Driver Developer Documentation team.

closed 2018-10-24 by DOMARS