MicrosoftDocs/windows-driver-docs #959: Missing power relations information

baget, 2018-10-22:
Hi
according to Doron Holon from MSFT in this thread:
https://community.osr.com/discussion/290710/ioregisterplugnplaynotification-does-not-work-for-eventcategorytargetdevicechange#latest

in order to use this notification, you need to define a power relations between the drivers.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a80d0e99-d153-889a-527f-d01ed47103cd
* Version Independent ID: 602d956a-9943-95c0-163c-d2e527add822
* Content: [Using PnP Target Device Change Notification](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/using-pnp-target-device-change-notification)
* Content Source: [windows-driver-docs-pr/kernel/using-pnp-target-device-change-notification.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/using-pnp-target-device-change-notification.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

duncanmacmichael, 2018-10-30:
Sure @EliotSeattle, I'll take a look at what the correct wording might be for clarifying this.

duncanmacmichael, 2018-10-31:
Hi @baget, thanks for waiting. I talked with the feature team and got a little more info. Basically, this is a PnP API that is not intended to notify drivers about power state changes. Instead, you should define a power relationship between devices. I've added a note to this effect to the page so this usage is clear. That should be live later today or early tomorrow. 

Thanks for helping to improve the docs!

closed 2018-10-31 by duncanmacmichael