MicrosoftDocs/windows-driver-docs #982: additional registry key details

pemj, 2018-10-30:
The document indicates that the user should proceed to technet.microsoft.com for more details on each registry key. However. when searching technet for each of these registry keys, the results tend to include a number of bug reports sorta related to this issue, and a link back to this article, but nothing else in terms of details about the particular keys.

Additionally, there are keys present in my registry's copy of HKLM\SYSTEM\CurrentControlSet\Services\i8042prt\Parameters that are not listed here, and the keys present in the base HKLM\SYSTEM\CurrentControlSet\Services\i8042prt are not documented anywhere as far as I can tell.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4be41871-7dfa-0222-936b-fd830396a269
* Version Independent ID: c132e39f-94de-5791-b0f8-636fa21aebb1
* Content: [PS/2 (i8042prt) driver - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/ps-2--i8042prt--driver)
* Content Source: [windows-driver-docs-pr/hid/ps-2--i8042prt--driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/hid/ps-2--i8042prt--driver.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

pemj, 2018-11-26:
@tedhudek , is this in your wheelhouse? I'd be happy to submit a PR with some changes if you can get me the contact info I need to do some research.

tedhudek, 2018-11-26:
Sorry @pemj I don't know anything about the above.  I'll poke around and see if I can find an owner.  Is there a specific question you have, or you're just looking for docs on all entries in that reg key?

pemj, 2018-12-06:
Mostly I'm trying to find more documentation on the registry keys around that area. I've been working with some other folks to debug an issue with PS/2 keyboard detection during boot, and I ended up running up against these docs in the process. Looks like the change you made addressed the technet link issue, so I'm comfortable closing the issue.

Thanks!

closed 2018-12-06 by pemj