MicrosoftDocs/windows-driver-docs #983: Provisioning failure - ERROR 11004 NoData

elad8a, 2018-10-30:
Although I can sucessfully ping target, I get this generic error message:

Provisioning log
Installing necessary components...
Failed operation: An error occurred while connecting from the remote machine.
Error: 11004 (NoData)
Error message: The requested name is valid, but no data of the requested type was found

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ce050ea3-d61e-830c-3038-f725eb4c29e1
* Version Independent ID: eb7fc65e-bcfe-dc86-3f8c-336d20eb0c86
* Content: [Provision a computer for driver deployment and testing (WDK 10) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/provision-a-target-computer-wdk-8-1)
* Content Source: [windows-driver-docs-pr/gettingstarted/provision-a-target-computer-wdk-8-1.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/provision-a-target-computer-wdk-8-1.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

user20112, 2018-11-08:
I fixed this by replaceing the name with the local ip of the machine. for some reason i couldnt get my network name to be recognized.
Goodluck

tedhudek, 2018-12-13:
THanks @user20112 for the report and then for circling back to report your fix. I just added the local IP suggestion to the page, should be live in a few minutes. Cheers!

closed 2018-12-13 by tedhudek