MicrosoftDocs/windows-driver-docs #985: Where are these programs located?

StingyJack, 2018-10-31:
Specifically for Windows 10 and Windows Server OS

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3b5d582a-57c8-8a72-ffcc-a2758ce75b96
* Version Independent ID: caab41cb-7bad-0618-683e-b6f4616545e3
* Content: [Debugging Using CDB and NTSD - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debugging-using-cdb-and-ntsd)
* Content Source: [windows-driver-docs-pr/debugger/debugging-using-cdb-and-ntsd.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugging-using-cdb-and-ntsd.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-11-02:
@StingyJack  - For information on how to download CDB/NTSD see this topic: [Debugging Tools for Windows (WinDbg, KD, CDB, NTSD)](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/). These tools don't really provide a UI, so another option to consider is using our newest debugger interface described in [Debugging Using WinDbg Preview](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debugging-using-windbg-preview).

closed 2018-11-02 by DOMARS

StingyJack, 2018-11-03:
@DOMARS - I meant where are they once installed. This was only a curiosity so I'll just wait for Cortana to decide it wants to _do_ something instead of just "working on it".

 I've used the new WinDbg preview and its not any better than the old one. The starting knowledge floor is still way too high for what should be common tasks (finding leaks, hang/crash reasons, etc), and there is 0 guidance with it,  leaving most who try to use it either more confused than before or unsure of how to interpret the results.