MicrosoftDocs/windows-driver-docs #987: Missing Param 1 value

jgyngell, 2018-11-01:
Missing param 1 value 0x404

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

jimtahu, 2018-12-05:
I hit this as well

jgyngell, 2019-05-10:
Any movement on this?

DOMARS, 2019-06-21:
@jgyngell - I have draft text prepared and it is being reviewed by the dev team for this parameter 1 value for bug check 0x1A.

DOMARS, 2019-07-01:
@jgyngell - The description for the  404 bug check 0x1A parameter 1 is live. Thanks for your interest in improving the stop code documentation.

closed 2019-07-01 by DOMARS