MicrosoftDocs/windows-driver-docs #988: Wrong error description for 1304

gerikintercede, 2018-11-01:
The error description above for 1305 belongs to 1304 as seen in the output from infverif

Output from C:\Program Files (x86)\Windows Kits\10\Tools\x64\infverif.exe
WARNING(1304) in C:\Dev\UMC\VSC\Debug\x64\MyIDVscDriver_Package\MyIDVscDriver.inf, line 98, target OS NTamd64.6.1: Found legacy AddReg operation using non-relative key (HKLM\SOFTWARE\Wow6432Node\Microsoft\Cryptography\Calais\SmartCards\Intercede Software Virtual Card).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 48275f87-4086-1ca5-a190-c741f29ad348
* Version Independent ID: 5481885d-3063-527b-a85e-6a898dbcb990
* Content: [INF Validation Errors and Warnings - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/inf-validation-errors-and-warnings)
* Content Source: [windows-driver-docs-pr/devtest/inf-validation-errors-and-warnings.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/inf-validation-errors-and-warnings.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-11-11:
@tedhudek, can you verify this?

tedhudek, 2018-11-13:
Updated the 130x values, which were all one higher than they should have been.  Thanks @gerikintercede

closed 2018-11-13 by tedhudek